Unity-Technologies/Unity-Robotics-Hub #355: Missing ROS Melodic Information on ROS setup

A-Ivan, 2022-02-10:
**Describe the bug**
In the ROS setup of the Pick-and-Place Tutorial: Part 0 > [Option B: Manual Setup](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/0_ros_setup.md#option-b-manual-setup), there is an explanation on how to setup ROS for both Melodic and Noetic.

Before the terminal commands for the Noetic version, the following title/phrase appears "ROS Noetic users should run:". I believe this similar phrase existed for the Melodic version setup, but it is missing now.

[image]

I believe the phrase "ROS Melodic users should run:" could be added to make things clearer for Melodic users.

at669, 2022-02-10:
Hey @A-Ivan! Looks like the line about Melodic is just before your box, without an extra line break. I'll go put it on the next line for consistency. Thanks!

A-Ivan, 2022-02-10:
You're welcome. I can see it now, thanks. Strange that without the line break my browser wouldn't show the line.

at669, 2022-02-11:
These documentation updates have been merged, and I'll close out this ticket. Thanks again!

closed 2022-02-11 by at669