Unity-Technologies/Unity-Robotics-Hub #357: Network Troubleshooting Update

A-Ivan, 2022-02-10:
**Describe the bug**
With the current version of the [ROS-TCP-Endpoint](https://github.com/Unity-Technologies/ROS-TCP-Endpoint) (0.7.0), the config folder and the params.yaml file have been removed.

I believe the troubleshooting notes in the [Pick-and-Place Tutorial: Part 2](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/2_ros_tcp.md#troubleshooting) page could be updated to reflect this. The following notes currently exist: 

"If Unity fails to find a network connection, ensure that the ROS IP address is entered correctly as the ROS IP Address in the RosConnect in Unity, and that the src/niryo_moveit/**config/params.yaml** values are set correctly."

"If the ROS TCP handshake fails (e.g. ROS-Unity server listening... printed on the Unity side but no ROS-Unity Handshake received on the ROS side), the ROS IP may not have been set correctly in the params.yaml file. Try running echo "ROS_IP: $(hostname -I)" > src/niryo_moveit/**config/params.yaml** in a terminal from your ROS workspace."

Information on the param file also appears in the [ROS–Unity Integration Networking](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/ros_unity_integration/network.md) page, as follows:

"On the ROS machine, these settings are set as a rosparam and will typically be set in a launch file like this or in a [param file](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/ROS/src/niryo_moveit/config/params.yaml) loaded by a launch file like this. The param file can also be loaded manually by running the rosparam load params.yaml command."

If I am not mistaken, I believe the launch file is not loading the params file anymore, the launch file already contains the IP and port data.

I would also like to suggest to having the network troubleshooting page in a place easier to be found. I don't remember the "Pick-and-Place Tutorial: Part 2" is the only place that the specific network troubleshooting appears. It would be helpful to also have it in the  [ROS-Unity Integration](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/ros_unity_integration/setup.md) or the [ROS–Unity Integration Networking](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/ros_unity_integration/network.md) document.

at669, 2022-02-10:
Thanks for the feedback! I'll be tracking these suggestions in an internal ticket and will pull any related updates in with #355.

[Ticket#: AIRO-1709]

at669, 2022-02-11:
These documentation updates have been merged, and I'll close out this ticket. Thanks!

closed 2022-02-11 by at669