Unity-Technologies/Unity-Robotics-Hub #358: Turtlebot Navigation: ROS 1 vs ROS 2

benchun123, 2022-02-17:
Hey,

I am really interested in Unity and ROS. After running the tutorials by Unity-Technology, I can simulate uArm with ROS and turtlebot 3 with ROS 2. Then, I have a question: why we use ROS 2 for turtlebot not ROS 1?

Currently, I have a real turtlebot, which is controlled by ROS 1, but in Unity Robotics, I can only find turtlebot with ROS 2. So, If I want to develop the simulation on ROS 1, is it possible? If so, what do I need to do?

Following are some examples and discussion I have found:
https://github.com/Unity-Technologies/Unity-Robotics-Hub
https://forum.unity.com/threads/introducing-ros-2-support-and-the-amr-demo.1155119/#post-7410458
https://github.com/DynoRobotics/UnityRos2

sarah-gibson, 2022-02-24:
Hi @benchun123 

Our ros-tcp-connector package actually supports ROS 1 as well as ROS 2! I suggest you take a look at [this documentation](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/ros_unity_integration/setup.md) and try to switch to ROS 1 on your own! You will also need to replace the ROS 2 navigation packages with corresponding ROS 1 packages, of which there are plenty.

closed 2022-02-24 by sarah-gibson

Hrithik-verma, 2022-04-08:
Hi @sarah-gibson 
ros-tcp-connector package support ros1 that okay but we also want support of ros1 Navigation. Problem is its very difficult to understand how to adopt (https://github.com/Unity-Technologies/Robotics-Nav2-SLAM-Example) for ros1. Even a page on how to adopt ros2 navigation for Ros1 will work out. Please try and understand that unity ros is very new and there is hardly any other source on internet except Unity-Robotics-Hub git to understand it.