Unity-Technologies/Unity-Robotics-Hub #366: Failed to build WebGL using PickandPlaceProject

qianxf001, 2022-04-07:
Hi ,

I follow the tutorials to set PickAndPlaceProject ,and build WebGL.
tutorials  link: https://github.com/Unity-Technologies/Unity-Robotics-Hub/tree/main/tutorials/pick_and_place.
After set "Pick-and-Place Tutorial: Part 2", I tried to buildWebGL but get error.
Could you help to check the error info if I missed some config?

**Environment Details:**
OS version: Ubuntu18.04
Unity Hub: 3.0.0-beta.7
Unity Editor Version: 2021.2.18f1c1
Target Platform:WebGl

**Console errors:**
Assets/Scripts/SourceDestinationPublisher.cs(5,38): error CS0234: The type or namespace name 'ROSGeometry' does not exist in the namespace 'Unity.Robotics.ROSTCPConnector' (are you missing an assembly reference?)
Assets/Scripts/SourceDestinationPublisher.cs(6,22): error CS0234: The type or namespace name 'UrdfImporter' does not exist in the namespace 'Unity.Robotics' (are you missing an assembly reference?)
Assets/Scripts/TrajectoryPlanner.cs(7,38): error CS0234: The type or namespace name 'ROSGeometry' does not exist in the namespace 'Unity.Robotics.ROSTCPConnector' (are you missing an assembly reference?)
ssets/Scripts/SourceDestinationPublisher.cs(29,5): error CS0246: The type or namespace name 'UrdfJointRevolute' could not be found (are you missing a using directive or an assembly reference?)
Assets/Scripts/SourceDestinationPublisher.cs(32,5): error CS0246: The type or namespace name 'ROSConnection' could not be found (are you missing a using directive or an assembly reference?)
Assets/Scripts/TrajectoryPlanner.cs(42,5): error CS0246: The type or namespace name 'ROSConnection' could not be found (are you missing a using directive or an assembly reference?)
Error building Player because scripts had compiler errors
Build completed with a result of 'Failed' in 2 seconds (2200 ms)
UnityEngine.GUIUtility:ProcessEvent (int,intptr,bool&) (at /home/bokken/buildslave/unity/build/Modules/IMGUI/GUIUtility.cs:189)
UnityEditor.BuildPlayerWindow+BuildMethodException: 7 errors
  at UnityEditor.BuildPlayerWindow+DefaultBuildMethods.BuildPlayer (UnityEditor.BuildPlayerOptions options) [0x002cc] in /home/bokken/buildslave/unity/build/Editor/Mono/BuildPlayerWindowBuildMethods.cs:193 
  at UnityEditor.BuildPlayerWindow.CallBuildMethods (System.Boolean askForBuildLocation, UnityEditor.BuildOptions defaultBuildOptions) [0x00080] in /home/bokken/buildslave/unity/build/Editor/Mono/BuildPlayerWindowBuildMethods.cs:94 
UnityEngine.GUIUtility:ProcessEvent (int,intptr,bool&) (at /home/bokken/buildslave/unity/build/Modules/IMGUI/GUIUtility.cs:189)

I have tried two solution which got from github:
**Solution1**
Following link:https://github.com/Unity-Technologies/ROS-TCP-Connector/blob/main/ROSGeometry.md
 Comment out "using Unity.Robotics.ROSTCPConnector"，which could slove the above error about ROSGeometry/UrdfImporter/UrdfJointRevolute,But still have the error about ROSConnection 
**New Console errors:**
Assets/Scripts/SourceDestinationPublisher.cs(32,5): error CS0246: The type or namespace name 'ROSConnection' could not be found (are you missing a using directive or an assembly reference?)
Error building Player because scripts have compile errors in the editor
Build completed with a result of 'Failed' in 1 seconds (829 ms)
UnityEngine.GUIUtility:ProcessEvent (int,intptr,bool&) (at /home/bokken/buildslave/unity/build/Modules/IMGUI/GUIUtility.cs:189)
UnityEditor.BuildPlayerWindow+BuildMethodException: Error building Player because scripts have compile errors in the editor
  at UnityEditor.BuildPlayerWindow+DefaultBuildMethods.BuildPlayer (UnityEditor.BuildPlayerOptions options) [0x002cc] in /home/bokken/buildslave/unity/build/Editor/Mono/BuildPlayerWindowBuildMethods.cs:193 
  at UnityEditor.BuildPlayerWindow.CallBuildMethods (System.Boolean askForBuildLocation, UnityEditor.BuildOptions defaultBuildOptions) [0x00080] in /home/bokken/buildslave/unity/build/Editor/Mono/BuildPlayerWindowBuildMethods.cs:94 
UnityEngine.GUIUtility:ProcessEvent (int,intptr,bool&) (at /home/bokken/buildslave/unity/build/Modules/IMGUI/GUIUtility.cs:189)

**Solution2**
Following link:https://github.com/Unity-Technologies/Unity-Robotics-Hub/issues/215
In Unity's Project window, right click on the package ROS TCP Connector, and select open containing folder. This should open an OS folder called PackageCache.
Do the same for your project's Assets folder. This should open your project's root folder.
Within the root folder , open the Packages folder.
Move the folder com.robotics.ros-tcp-connector@<miscellaneous-characters> from your PackageCache window into the Packages folder. This will enable you to make and save changes to the package.
Now, in Unity, you should be able to go into the Packages/com.unity.robotics.ros-tcp-connector/Runtime/Unity.Robotics.ROSTCPConnector.asmdef file and tick the "WebGL" checkbox .
Then  got an error info:
Unapplied import settings for 'Packages/com.unity.robotics.ros-tcp-connector/Runtime/Unity.Robotics.ROSConnector.asmdef'
Screenshot:
[image]

Ps. do tell me if more details is required. I am relatively new to Unity and posting issues on GitHub, thus I might have missed out on details/ not know the conventions very well. Thank you!

qianxf001, 2022-04-07:
build setting screenshot:
[image]
Playerseting screenshot:
[image]

qianxf001, 2022-04-07:
URDF-Importer and ROS-TCP-Connector version:
[image]

starlprd, 2022-12-29:
I have run into same problem as well. Anyone got solution for it. I did override the plugin from packages to build the app but it doesn't connect after building. Does this not support WebGL?

mshpm, 2023-01-27:
I am facing the same issue while running in WebGL. Anyone got the solution?

learning-github-zgy, 2023-11-17:
Hello, I have the same problem as you, have you solved it?

vmarquim, 2024-10-07:
I have currently the same problem with the WebGL build. Did somebody solve it?