Unity-Technologies/Unity-Robotics-Hub #373: Pick-and-place Part2 Connection failed

Nekomiya-fubuki, 2022-05-19:
**Describe the bug**
I followed the pick-and-place tutorial step by step,and I finished the Part0 and Part1.But when I continue to do the Part2,I can;t
get a connection between ros and unity.I think I set the right ROS ip

**Screenshots**
ROS worked fine, but Unity couldn't connect
[image]

[image]

**Environment (please complete the following information, where applicable):**
- Unity Version: 2020.3.30f1c1
- ROS Version:18.04 with Oracle VM VirtualBox

sergiosolorzano, 2022-12-17:
Hi @Nekomiya-fubuki I'm having the same problem, VirtualBox VM and Ubuntu 18.04, ROS works but Unity won't connect. Did you solve this and can share please?

MischaRo, 2023-06-14:
I am having the exact same problem. 
Is there any progress on this?

MischaRo, 2023-06-14:
Ok, for anybody having this issue, I did two things to fix this:
As stated in the troubleshooting  for the network tutorial, I set exceptions for the Ports 10000 and 5005 as well as for the unity app in the firewall

https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/ros_unity_integration/network.md#open-port-on-windows-firewall

Another thing that I did was to make sure I could actually  ping my virtual ubunto machine,  which I could not.
To fix this, I would had to change the network settings of the Virtual Box to Bridge Adapter. 

https://askubuntu.com/questions/416441/cant-ping-virtualbox-server-operating-system

After this was done, I had to check and set up the new IP in the ubuntu vm and Unity could connect.