Unity-Technologies/Unity-Robotics-Hub #375: UDP support request

palwanishan, 2022-06-01:
As I know currently communication between ROS and Unity is done by TCP which is exactly same as ROS communication itself. Partially, ROS supports UDP as well but not widely used or supported. I would like to send and get bigger data such as LIDAR, RGB and Depth. Currently, it doesn't transfer any data other than first frame in most cases, even in the fast local network. 
My personal hypothesis is UDP might help for accelerating data transfer in such cases. 
Please, let us know if the issue is not because of the protocol, it will be helpful for most of us. I saw many of this kind questions in the youtube video comments which are related to this repository.

YW0330, 2023-03-18:
Hi, @palwanishan 

Do you have any solution for this issue?
Since I need to publish the depth image, I also want to find a UDP protocol between ROS and Unity.

Thank you in advence.

missxa, 2024-01-29:
+1 for UDP support