Unity-Technologies/Unity-Robotics-Hub #384: TCP connection failed in Unity in Pick and Place Tutorial—part 2  (using docker)

owlmoo, 2022-10-27:
**Describe the bug**
I follow the instruction to set up the docker and unity. However, when I hit play, I cannot build the connection. 
In unity, it reports "Connection to 127.0.0.1:10000 failed" and "The remote host forcibly closed an existing connection".

[image]

“远程主机强迫关闭了一个现有的连接” means "The remote host forcibly closed an existing connection".

**To Reproduce**
In the ros side, I just source the setup.bash and run the part2.launch(roslaunch niryo_moveit part_2.launch).
In the unity side, I just follow the instuction and hit play.
[image]
[image]

**Console logs / stack traces**

`Connection to 127.0.0.1:10000 failed - System.IO.IOException: Unable to write data to the transport connection: 远程主机强迫关闭了一个现有的连接。
UnityEngine.Debug:Log (object)
Unity.Robotics.ROSTCPConnector.ROSConnection/<ConnectionThread>d__115:MoveNext () (at Library/PackageCache/com.unity.robotics.ros-tcp-connector@c27f00c6cf/Runtime/TcpConnector/ROSConnection.cs:854)
System.Threading._ThreadPoolWaitCallback:PerformWaitCallback ()
`

**Environment (please complete the following information, where applicable):**
- Unity Version: Unity 2020.3.35f1
- Unity machine OS + version: Windows 10
- ROS machine OS + version: Ubuntu 18.04, ROS Melodic
- ROS–Unity communication: Docker
- Branch or version: v0.7.0

sergiosolorzano, 2022-12-17:
Hi @owlmoo  I could run ubuntu connection on v.18.04 but when I upgraded to 20.04 it fails, did you have to install ROS in your system too or only followed the commands shown in the [ros manual installation](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/0_ros_setup.md#option-b-manual-setup)?

When I try command:
sudo apt-get install python-pip ros-melodic-robot-state-publisher ros-melodic-moveit ros-melodic-rosbridge-suite ros-melodic-joy ros-melodic-ros-control ros-melodic-ros-controllers ros-melodic-tf2-web-republisher

I get:
sudo apt-get install python-pip ros-melodic-robot-state-publisher ros-melodic-moveit ros-melodic-rosbridge-suite ros-melodic-joy ros-melodic-ros-control ros-melodic-ros-controllers ros-melodic-tf2-web-republisher

owlmoo, 2022-12-19:
@sergiosolorzano Hi, I have this problem too, ros noetic support python3, and I just re-link the python interpreter using : 
sudo ln -s /usr/bin/python3 /usr/bin/python