Unity-Technologies/Unity-Robotics-Hub #391: After running ros2 run ros_tcp_endpoint default_server_endpoint --ros-args -p ROS_IP:=0.0.0.0, nothing happens.

gkgkgkgk, 2023-03-01:
I have ROS2 configured on a Docker machine, and when I get up to the step of running this command:
`ros2 run ros_tcp_endpoint default_server_endpoint --ros-args -p ROS_IP:=0.0.0.0` the console freezes and nothing happens. 

I cant seem to quit out of the process either, and I don't the `[INFO] [1603488341.950794]: Starting server on 192.168.50.149:10000` confirmation as described in the documentation. When I run unity and the example publisher code, I get this error: 

`Connection to 127.0.0.1:10000 failed - System.ArgumentException: Addresses 127.0.0.1 (IPv4) and ::0 (IPv6) are unspecified addresses. You cannot use them as target address.`. 

I am assuming I get this error because of the initial failure I described above, although I could be wrong. Any ideas?

Miftahur92, 2024-01-24:
HI. I a, facing the same problem. Did you solve this problem?

Murage41, 2025-01-14:
Create an empty game object for ros connection then add in the inspector the ros connection script. or just add the ros connection prefab from project window