Unity-Technologies/Unity-Robotics-Hub #392: Run roslaunch niryo_moveit part_2.launch   but  process has died [pid 10010, exit code 127

bi1mena, 2023-03-25:
**Describe the bug**
In Pick-and-Place Tutorial Part 2: ROS–Unity Integration
I Run roslaunch niryo_moveit part_2.launch   
but See error
process has died [pid 10010, exit code 127

**To Reproduce**
Steps to reproduce the behavior:
1. Go to my workspace
2. run catkin_make && source devel/setup.bash
3. Run roslaunch niryo_moveit part_2.launch
4. See error

**Console logs / stack traces**
process[server_endpoint-2]: started with pid [10010]
/usr/bin/env: ‘python’: No such file or directory
[server_endpoint-2] process has died [pid 10010, exit code 127, cmd /home/lz/ROS/src/ros_tcp_endpoint/src/ros_tcp_endpoint/default_server_endpoint.py --wait __name:=server_endpoint __log:=/home/lz/.ros/log/fa595246-cb21-11ed-b866-086ac5a336a7/server_endpoint-2.log].
log file: /home/lz/.ros/log/fa595246-cb21-11ed-b866-086ac5a336a7/server_endpoint-2*.log
[server_endpoint-2] restarting process

**Expected behavior**
I hope to get the same result as in the tutorial.

**Screenshots**
[image]

**Environment (please complete the following information, where applicable):**
- Unity Version: [e.g. Unity 2020.3.11f1]
- Unity machine OS + version: [e.g. Windows 10]
- ROS machine OS + version: [e.g. WSL Ubuntu 20.04, ROS Noetic]

**Additional context**
Hope my problem can be solved, thank you very much

aixuexidexiaoyuyu, 2023-08-04:
same error, did you solve it?

Iris-Orion, 2023-09-14:
> same error, did you solve it?

Hi, here is how i solved it :
.  you should open /ROS/Unity-Robotics-Hub/tutorials/pick_and_place/ROS/src/ros_tcp_endpoint/src/ros_tcp_endpoint/default_server_endpoint.py
.  i open this .py in vim and change the first line #!/usr/bin/env python to #!/usr/bin/env python3
this works for me and i think the main reason is you should use python3 in ros noetic instead of python

learning-github-zgy, 2023-11-20:
I have the same problem as you. Have you solved it?

rerdcn, 2024-08-16:
I fix it.
Open default_server_endpoint.py change first line
" #!/usr/bin/env python"             To               "#!/usr/bin/env python3"