Unity-Technologies/Unity-Robotics-Hub #400: ERROR: failed to solve: process "/bin/sh -c sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys

suerich93, 2023-06-21:
**ERROR: failed to solve: process "/bin/sh -c sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys **
When trying to run the Unity tutorial Pick and Place on Windows with WSL 2 and Docker, I receive the issue ERROR: 

RUN sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys F42ED6FBAB17C654:
#0 0.340 Warning: apt-key output should not be parsed (stdout is not a terminal)
#0 0.352 Executing: /tmp/apt-key-gpghome.1yzQu92SDH/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-keys F42ED6FBAB17C654
#0 60.52 gpg: keyserver receive failed: Connection timed out

The issue does not occur when using Tutorial Ros Unity integrator. I assume some issue with the Docker key. 

**To Reproduce**
Steps to reproduce the behavior:
Follow the guidelines of the tutorial with Windows 10, WSL 2 Ubuntu, Unity 22.3.2f1

**Console logs / stack traces**

Please wrap in [triple backticks (```)](PS D:\Unity\2022.3.2f1\Unity-Robotics-Hub\tutorials\pick_and_place> docker build -t foxy -f docker/Dockerfile .
[+] Building 61.2s (6/19)
 => [internal] load .dockerignore                                                                                  0.0s
 => => transferring context: 2B                                                                                    0.0s
 => [internal] load build definition from Dockerfile                                                               0.0s
 => => transferring dockerfile: 1.25kB                                                                             0.0s
 => [internal] load metadata for docker.io/library/ros:melodic-ros-base                                            0.6s
 => CACHED [ 1/15] FROM docker.io/library/ros:melodic-ros-base@sha256:7e6fbcc36b05b7b63adec64f29ed7b397779f11123d  0.0s
 => [internal] load build context                                                                                  0.0s
 => => transferring context: 63.88kB                                                                               0.0s
 => ERROR [ 2/15] RUN sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys F42ED6FBAB17C654              60.6s
------
 > [ 2/15] RUN sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys F42ED6FBAB17C654:
#0 0.340 Warning: apt-key output should not be parsed (stdout is not a terminal)
#0 0.352 Executing: /tmp/apt-key-gpghome.1yzQu92SDH/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-keys F42ED6FBAB17C654
#0 60.52 gpg: keyserver receive failed: Connection timed out
------
Dockerfile:3
--------------------
   1 |     FROM ros:melodic-ros-base
   2 |
   3 | >>> RUN sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys F42ED6FBAB17C654
   4 |
   5 |     RUN sudo apt-get update && sudo apt-get install -y vim iputils-ping net-tools python-pip ros-melodic-robot-state-publisher ros-melodic-moveit ros-melodic-rosbridge-suite ros-melodic-joy ros-melodic-ros-control ros-melodic-ros-controllers ros-melodic-tf2-web-republisher dos2unix
--------------------
ERROR: failed to solve: process "/bin/sh -c sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys F42ED6FBAB17C654" did not complete successfully: exit code: 2
PS D:\Unity\2022.3.2f1\Unity-Robotics-Hub\tutorials\pick_and_place>) to make it easier to read.

**Expected behavior**
To receive a Docker image build

**Environment (please complete the following information, where applicable):**
- Unity Version: [Unity 22.3.2f1]
- Unity machine OS + version: [Windows 10]
- ROS machine OS + version: [WSL 2 Ubuntu]
- ROS–Unity communication: [Docker Desktop trying to be built in PS]

jurgispods, 2023-11-10:
Try replacing the line

```
RUN sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys F42ED6FBAB17C654
```
with

```
RUN sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list' >/dev/null 2>&1 \
    && sudo apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
```

BrundenW, 2024-03-07:
> Try replacing the line
>
> ```
> RUN sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys F42ED6FBAB17C654
> ```
>
> with
>
> ```
> RUN sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu [...]

This pretty much worked for me, but I had to combine the two bits of code to get this instead
`RUN sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list' >/dev/null 2>&1 \
    && sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys F42ED6FBAB17C654`