Unity-Technologies/Unity-Robotics-Hub #401: I want to use unity to connect to multiple ROS systems, but it doesn't seem to work

wenhaogan9898, 2023-07-06:
I want to use unity to connect to multiple ROS systems, but the code in ROSConnection to create the ros connection instance is static, which means it can always connect to only one ROS system. I wonder if there is another way to connect multiple ROS systems (different IP and port) at the same time.

`        
public static void DrawConnectionArrows(bool withBar, float x, float y, float receivedTime, float sentTime, bool isPublisher, bool isSubscriber, bool hasError);
        public static Color GetConnectionColor(float elapsedTime, bool hasConnection, bool hasError);
        public static ROSConnection GetOrCreateInstance();
        public static bool IPFormatIsCorrect(string ipAddress);
        public static void SetIPPref(string ipAddress);
        public static void SetPortPref(int port);
`

FGJoseMiguel-AIST, 2023-09-19:
I want to do the same, but I don't have any idea how. Do you already do it?

FGJoseMiguel-AIST, 2023-11-09:
You have to define your ROS Connection like a public variable, after that call your subscribers or publishers using the same variable. In your scene you have to define 2 empty objects like ROS-1 and ROS-2, add the component named ROS Connection. When you finish that only rest drag the ROS-1 and ROS-2 objects to your variables.