Unity-Technologies/Unity-Robotics-Hub #409: RosConnection.Update() spiking to >180ms execution. This thread-blocking causes bad frame spikes when Playing.

cokumura2019, 2023-11-11:
**Describe the bug**
I am getting massive stutters when streaming images from ROS. When using the Editor profiler, it's apparent that the RosConnection.Update() invocation is blocking the main thread for >180ms. I am also using 

Note: I am using 5, 720x720 images (~45fps per cam)

**To Reproduce**
Steps to reproduce the behavior:
1. Start the Unity TCP endpoint specifying 127.0.0.1 as the IP address (via the bash terminal)
2. Set up 5 visualizers using the Unity Visualization package, each connecting to a different ROSTopic (so each visualizer window is subscribed to a different Image topic)
3. Open up the Unity Profiler window
4. Observe the lag spikes!

**Console logs / stack traces**
No errors thrown. 

**Expected behavior**
I guess I would expect that the Update() invocation shouldn't take so long to compute. 90% of the other frames that render the 5 camera views are smooth. However, when the frame lag spike occurs, it's very noticeable so I'd expect that any delay on the Update() function > 30ms is concerning. 

**Screenshots**
Screenshot of frame showing massive compute time for RosConnection.Update():
[image]

Screenshot of frame showing normal compute time for RosConnection.Update()
[image]

**Environment (please complete the following information, where applicable):**
- Unity Version:  Unity 2022.3.10f1
- Unity machine OS + version: Ubuntu 20.04
- ROS machine OS + version: Ubuntu 20.04, ROS Noetic
- ROS–Unity communication: localhost via Unity-ROS TCP Endpoint (C++ version)
- Branch or version: [v0.7.0 for the Unity-ROS TCP Connector for Unity]   [v0.7.0 for the Unity-ROS visualization package]   [v0.7.0 for Unity-ROS TCP endpoint]

**Additional context**
These delay spikes occur relatively frequently (>2 times per second)

cokumura2019, 2023-11-12:
One other odd thing I noted:
- When I made only 1 ROS topic subscription, a large lag spike was still present but occurred predictably around every 3 seconds, with smaller lag spikes every second.

zAle711, 2023-11-17:
I noticed the same thing, please leave an update if you find a fix!