Unity-Technologies/Unity-Robotics-Hub #418: Connection and fast Disconnected and then recurrent both, the terminal info ModuleNotFoundError: No module named 'unity_robotics_demo_msgs.unity_robotics_demo_msgs_s__rosidl_typesupport_c'

shiqikuangsan2, 2024-05-24:
[INFO] [1716546449.579505198] [UnityEndpoint]: Connection from 172.16.46.161
[INFO] [1716546449.691776483] [UnityEndpoint]: RegisterSubscriber(/tf, <class 'tf2_msgs.msg._tf_message.TFMessage'>) OK
[INFO] [1716546449.695356486] [UnityEndpoint]: Disconnected from 172.16.46.161
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_generator_py/import_type_support_impl.py", line 46, in import_type_support
    return importlib.import_module(module_name, package=pkg_name)
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1004, in _find_and_load_unlocked

ModuleNotFoundError: No module named 'unity_robotics_demo_msgs.unity_robotics_demo_msgs_s__rosidl_typesupport_c'

shiqikuangsan2, 2024-05-25:
Sorry，I have already solved this problem. It's the customer interface CMakeList problem

DongzhenHuangfu, 2024-07-11:
> Sorry，I have already solved this problem. It's the customer interface CMakeList problem

Could you please show me how you fix this problem?

lifuboshiqikuangsan, 2024-07-12:
https://blog.csdn.net/weixin_44566432/article/details/133941241

lifuboshiqikuangsan, 2024-07-12:
> https://blog.csdn.net/weixin_44566432/article/details/133941241

you can solve this problem by follow this paper