Unity-Technologies/Unity-Robotics-Hub #433: Build Fail - Unable to Build Unity Scene

vstavros02, 2025-06-13:
I github installed the URDF Importer from this link “[GitHub - Unity-Technologies/URDF-Importer: URDF importer](https://github.com/Unity-Technologies/URDF-Importer.git)” the 0.5.2 version and I’m experiencing experiencing issues. I’ve imported my URDF successfully on Unity however it fails to build, I believe the windows assimp.dll file seems to be causing issues as I’m trying to build under android (as I’ve ready this is what I should do in order to connect the scene to my Meta Quest 2 headset). Bellow is the terminal error.

[image]

**Environment:**
URDF Importer 0.5.2
Unity Version: Unity 6.0 (60000.0.47f1)
Unity machine: Working on Linux dual boot system
ROS machine OS + version: [e.g. Ubuntu 24.04, ROS2 Jazzy (however not using it for this)

ipa-jsk, 2025-06-18:
Check #169

salexiou, 2025-06-19:
2025/06/19 09:01:25.874 15055 15076 Error Unity InvalidOperationException: Cannot determine OS-specific implementation.
2025/06/19 09:01:25.874 15055 15076 Error Unity   at Assimp.Unmanaged.UnmanagedLibrary.GetPlatform () [0x00000] in <00000000000000000000000000000000>:0 

I get this error when building. ROS seems to be working but URDF is not.